jkiv/shapool-core #25: Cool! - Can we know what is the speed?

akshar001, 2021-07-31:
Hey, this is a very cool project. 
Can we know what is the speed of this?

akshar001, 2021-07-31:
I think i am the first one to comment on this!

jkiv, 2021-08-04:
It'll compute a hash every 64 cycles per core, so it depends on your clock speed.

This was designed on/for the iCE40HX8K-BG121 which currently can only support 2 cores at 30MHz. So roughly 0.9~1 MH/s.

However, the logic is deep. It performs a hash round every cycle. You can likely go faster with more pipelining at the expense of space on the FPGA. Not sure yet which choice would come out on top.

Some other optimizations, like putting the message in block RAM, could save some room and potentially bring the number of cores to 4 -- so 1.9~2 MH/s.

The [icepool](http://github.com/jkiv/icepool-board) project allows to cluster multiple iCE40HX8K's. One motherboard with 16 FPGAs would be about 14~15 MH/s.

In conclusion, not very fast.

closed 2021-08-04 by jkiv

akshar001, 2021-08-04:
Per core how many LUTs? for ICE40?

akshar001, 2021-08-04:
And if we have 50mhz clock then it would be more speed right?

jkiv, 2021-08-04:
The iCE40s have 7680 "logic cells." The cores share these with the I/O logic (2xSPI, control signals, etc.) so hard to say exactly. I think I can fit 3 cores and I/O into 7680 (if I recall correctly) but the logic divides the work up into powers of 2, so 3 doesn't work well.

And yeah, more cycles / second means more hashes / second.

akshar001, 2021-08-04:
you know about the esp32?

akshar001, 2021-08-04:
https://www.efinixinc.com/products-titanium.html

Ti35